google-deepmind/dm_control #454: Hash added to stl file name with to_xml_string()

curtiscjohnson, 2024-03-25:
Hello,

I am using pymjcf to programatically generate some complex Mujoco simulations. I have encountered an odd behavior where extra letters and numbers (appears to be a hash of some sort) are added to the file attribute of the mesh asset, which causes the file to not be found when loading the xml for simulation (e.g. meshed_cube.stl added via pymjcf appears in xml string as meshed_cube-64221cea0ccda8a91d52148f53eacf8bef9392a9.stl). Is this a bug or am I using pymjcf incorrectly?

Here is a MWE to demonstrate the issue: [mwe.zip](https://github.com/google-deepmind/dm_control/files/14746149/mwe.zip)

I am using the following packges installed via pip on Ubuntu 20.04 with Python 3.8.10:
- dm_control v1.0.16
- mujoco v3.1.3

hubernikus, 2024-04-15:
Same issue occurs when direclty loading assets from file using:
```
from dm_control import mjcf
self.mjcf_root_element = mjcf.from_file("data/scene.xml")
```

with an asset:
```
<asset>
   <mesh name="mymesh" file="mymesh.stl" />
</asset>
```

Results in an error:
```
ValueError: Error: resource not found via provider or OS filesystem: 'mymesh-df73f487580c2b9c9a3f6bac7f8d1e5b89374ba6.stl'
```

yuvaltassa, 2024-04-15:
This is not a bug, but a somewhat annoying thing that pymjcf does for reasons that are not relevant here. You can do some post processing to remove them like:
```py
from lxml import etree

print('Remove class="/".')
  xml_string = model.to_xml_string('float', precision=3, zero_threshold=1e-7)
  root = etree.XML(xml_string,
                   etree.XMLParser(remove_blank_text=True))
  default_elem = root.find('default')
  root.insert(3, default_elem[0])
  root.remove(default_elem)

  print('Remove hashes from filenames.')
  meshes = [mesh for mesh in root.find('asset').iter() if mesh.tag == 'mesh']
  for mesh in meshes:
    name, extension = mesh.get('file').split('.')
    mesh.set('file', '.'.join((name[:-41], extension)))

  print('Get string from lxml.')
  xml_string = etree.tostring(root, pretty_print=True)
  xml_string = xml_string.replace(b' class="/"', b'')
```

closed 2024-04-15 by yuvaltassa

hubernikus, 2024-04-15:
Alternatively use the function `mjcf.export_with_assets(..)`:
```
mjcf.export_with_assets(torso.mjcf_model, self.mjcf_root, "mwe.xml")
```

That will not require any post processing, but will create duplicates of your mesh with the correct name in the specified folder.