google-deepmind/dm_control #459: Unable to load an .xml with bodies which have "user" parameter using "mjcf.from_path()"

barikata1984, 2024-04-06:
Hi,  I'm a student and I'm trying to use MuJoCo for simulating rigid body dynamics.  

I'm looking for some help with setting "user" attribute in "body" elements in .XML files. Reading [this section](https://mujoco.readthedocs.io/en/stable/modeling.html#user-parameters) and referring to [these lines](https://github.com/google-deepmind/mujoco/blob/7d3b662d5b3e3c3f6cf9ed3f17f3ab4dda4923f3/test/user/user_objects_test.cc#L1188), I tried to set a user parameter as follows:  

```XML 
<mujoco model="target">
  <size nuser_body="1"/>
  <worldbody>
    <body name="object" user="1.28">
      <geom name="geom" type="box" density="1000" size=".16 .32 .64" rgba="1 0 0 1"/>
    </body>minimal
  </worldbody>
</mujoco> 
``` 

But I got the following error: 
``` 
Traceback (most recent call last):
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 210, in _parse_children
    mjcf_child = mjcf_element.add(xml_child.tag, **attributes)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 630, in add
    return self.insert(element_name, position=None, **kwargs)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 655, in insert
    new_element = _make_element(child_spec, self, attributes=kwargs)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 128, in _make_element
    return _ElementImpl(spec, parent, attributes)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 185, in __init__
    self._check_valid_attribute(attribute_name)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 534, in _check_valid_attribute
    raise AttributeError(
AttributeError: 'user' is not a valid attribute for <body>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 104, in from_path
    return _parse(xml_root, escape_separators,
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 169, in _parse
    _parse_children(xml_root, mjcf_root, escape_separators)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 219, in _parse_children
    _parse_children(xml_child, mjcf_child, escape_separators)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 216, in _parse_children
    raise err_type(
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/parser.py", line 210, in _parse_children
    mjcf_child = mjcf_element.add(xml_child.tag, **attributes)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 630, in add
    return self.insert(element_name, position=None, **kwargs)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 655, in insert
    new_element = _make_element(child_spec, self, attributes=kwargs)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 128, in _make_element
    return _ElementImpl(spec, parent, attributes)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 185, in __init__
    self._check_valid_attribute(attribute_name)
  File "/home/atsushi/miniconda3/envs/rigid_body_manip_se3/lib/python3.9/site-packages/dm_control/mjcf/element.py", line 534, in _check_valid_attribute
    raise AttributeError(
AttributeError: Line 4: error while parsing element <body>: 'user' is not a valid attribute for <body>
```  

Does `mjcf` support .XML files with "body" elements with a "user" parameter in the first place? If it supports, which part should I change in the XML string above?

Thanks in advance,

bakehamster, 2024-04-07:
Hey,
I am just a student as well not an expert, but I tried out your model in a standalone mujoco viewer. It loaded without any problem.  
(＾＾)ｂ

bakehamster, 2024-04-07:
Hey, I am a student as well not an expert. I have a similar problem just like you in Issue #457. I have not received any answers yet. But I solved it pretty unprofessionally, it worked tho. Something like this:
```
xml = env_mjcf.to_xml_string()
xml = xml.replace("<flexcomp ", '<flexcomp dim="3"', 1)
model = mujoco.MjModel.from_xml_string(xml)
```

yuvaltassa, 2024-04-07:
Hi both.

You probably just need to update mjcf/schema.xml?

Please try and if that solves your problem, send a PR?

Cheers

bakehamster, 2024-04-08:
Hey guys,

On the github version [mjcf/schema.xml](https://github.com/google-deepmind/dm_control/blob/main/dm_control/mjcf/schema.xml) they are not defined as valid attributes. After manually added the dim attribute to the flexcomp element, it solved my issue.

Thanks for the help.

Regards

barikata1984, 2024-05-28:
For future visitors looking for a workaround until PR #468 is merged, `custom/numeric` field can be used without the PR like:
```
  <custom>
    <numeric name="your_parameter" size="1" data="3.14"/>
  </custom>
```