google-deepmind/dm_control #462: mjcf/schema.xml is missing eulerdamp as flag attribute

JeyRunner, 2024-04-11:
Hi, I noticed that I get the following error when parsing my XML file with mjcf:
```
error while parsing element <flag>: 'eulerdamp' is not a valid attribute for <flag>
```
I guess that can be fixed by adding the following line in `/mjcf/schema.xm`:
```xml
        <element name="flag">
          <attributes>
             <attribute name="eulerdamp" type="keyword" valid_values="disable enable"/>
             <!-- ... -->
```
See https://mujoco.readthedocs.io/en/stable/XMLreference.html#option-flag .

yuvaltassa, 2024-04-11:
Yes. Please try it locally and if it works for you, send us a PR?

kevinzakka, 2024-09-03:
Fixed by b3f76589bbe766be47dbb32f19bb7b8c0aa98014.

closed 2024-09-03 by kevinzakka