google-deepmind/dm_control #464: Headless rendering on Mac?

dghw, 2024-04-12:
I'm currently trying to run something on a headless Mac server that is hanging due to a call to Physics.render()

Relevant code excerpt:
```
from dm_control.suite import base

class MyTask(base.Task):

  ...

  def get_observation(self, physics):
    image = physics.render(...)  
    ...

my_task.get_observation(self, physics)

```

I face the same issue if I try to follow the 'getting started' steps outlined here: https://github.com/google-deepmind/dm_control/tree/main/dm_control/mujoco

From a few existing issues and the description on the main readme.md, I can't quite tell if headless rendering on a MacOS system is supported or not - it seems like it might only be supported for Nvidia systems. If so, are there any known workarounds?

yuvaltassa, 2024-04-15:
@kevinzakka you render on Mac, right? Any thoughts?

saran-t, 2024-04-15:
In theory the CGL code path allows an OpenGL context to be created without a window. In practice the call to create a CGL context hangs until a graphical login session for the current user is created. I don't believe this is actually documented anywhere (and the entire CGL API is marked deprecated by Apple several years ago).

I've spent several days in the past trying to figure out how to open a graphical login session via SSH but have come to a conclusion that this isn't possible (or at least not through any documented means).

For our CI server the "workaround" is to VNC into the login screen and log in to the CI user after each reboot. In other words, it doesn't work headlessly.