google-deepmind/dm_control #473: MJCF Parsing Error - Identifier Containing Separator

peterdavidfagan, 2024-05-23:
I really like using MJCF for configuring my robots and environments. Recently I tried to upgrade one of my packages to the `robot_descriptions` package which references the mujoco_menagerie robot models. 

When making this transition I encountered parsing errors related to the latest robot models in mujoco_menagerie containing separators, which is raised as an error in the following [section](https://github.com/google-deepmind/dm_control/blob/04ff055d0b449742a3e075af6c0a711c2edfae89/dm_control/mjcf/attribute.py#L255) of the dm_control code.

The error can be reproduced with the latest versions of mujoco, robot_descriptions and mujoco_menagerie

```python
import os 

from dm_control import mjcf
from robot_descriptions import panda_mj_description

PANDA_MJCF_PATH = os.path.join(panda_mj_description.PACKAGE_PATH, 'panda_nohand.xml')

franka_mjcf = mjcf.from_path(PANDA_MJCF_PATH)
```

Error:
```
raceback (most recent call last):
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 205, in __init__
    self._attributes[attribute_spec.name] = attribute_spec.type(
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 59, in __init__
    self._check_and_assign(value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 65, in _check_and_assign
    self._assign_from_string(new_value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 106, in _assign_from_string
    self._assign(string)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 256, in _assign
    raise ValueError(
ValueError: An identifier cannot contain a '/', as this is reserved for scoping purposes: got 'panda/visual'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 210, in _parse_children
    mjcf_child = mjcf_element.add(xml_child.tag, **attributes)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 630, in add
    return self.insert(element_name, position=None, **kwargs)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 655, in insert
    new_element = _make_element(child_spec, self, attributes=kwargs)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 124, in _make_element
    return _DefaultElement(spec, parent, attributes)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 217, in __init__
    raise err_type(  # pylint: disable=raise-missing-from
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 205, in __init__
    self._attributes[attribute_spec.name] = attribute_spec.type(
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 59, in __init__
    self._check_and_assign(value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 65, in _check_and_assign
    self._assign_from_string(new_value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 106, in _assign_from_string
    self._assign(string)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 256, in _assign
    raise ValueError(
ValueError: during initialization of attribute 'class' of element <default>: An identifier cannot contain a '/', as this is reserved for scoping purposes: got 'panda/visual'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 104, in from_path
    return _parse(xml_root, escape_separators,
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 169, in _parse
    _parse_children(xml_root, mjcf_root, escape_separators)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 219, in _parse_children
    _parse_children(xml_child, mjcf_child, escape_separators)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 219, in _parse_children
    _parse_children(xml_child, mjcf_child, escape_separators)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 216, in _parse_children
    raise err_type(
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 210, in _parse_children
    mjcf_child = mjcf_element.add(xml_child.tag, **attributes)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 630, in add
    return self.insert(element_name, position=None, **kwargs)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 655, in insert
    new_element = _make_element(child_spec, self, attributes=kwargs)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 124, in _make_element
    return _DefaultElement(spec, parent, attributes)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 217, in __init__
    raise err_type(  # pylint: disable=raise-missing-from
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 205, in __init__
    self._attributes[attribute_spec.name] = attribute_spec.type(
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 59, in __init__
    self._check_and_assign(value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 65, in _check_and_assign
    self._assign_from_string(new_value)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 106, in _assign_from_string
    self._assign(string)
  File "/Users/peterfagan/Library/Caches/pypoetry/virtualenvs/mujoco-robot-environments-_OvGmQF9-py3.10/lib/python3.10/site-packages/dm_control/mjcf/attribute.py", line 256, in _assign
    raise ValueError(
ValueError: Line 15: error while parsing element <default>: during initialization of attribute 'class' of element <default>: An identifier cannot contain a '/', as this is reserved for scoping purposes: got 'panda/visual'
```

yuvaltassa, 2024-05-23:
I think @erez-tom just dealt with a similar issue?

kevinzakka, 2024-05-23:
Can you try using escape_separators please?

peterdavidfagan, 2024-05-23:
> Can you try using escape_separators please?

Ah my bad on this, I don't know how I managed to miss the `escape_serparators` function argument. Thanks for the responses on this, hopefully this helps someone else who also managed to miss this argument.

closed 2024-05-23 by peterdavidfagan