google-deepmind/dm_control #486: Is there a way to get the .xml file of the locomotion soccer environment? Run environment in GPU?

StefanoBraghetto, 2024-07-23:
I want to create a project from this environment: [soccer](https://github.com/google-deepmind/dm_control/tree/main/dm_control/locomotion/soccer)
But I can't get with the .xml. My final objective is to run it in the GPU so I can run multiple instance of it at the same time. I was reading that this is possible with **mujoco_mjx** and I could load the environment doing:
```
mj_model = mujoco.MjModel.from_xml_path(
        (path / 'name_file.xml').as_posix())
```

please help!

sibocw, 2024-10-18:
I think #429 gives the answer, unless the developers have since decided to retrofit MJX?

kevinzakka, 2024-10-18:
@sibocw @StefanoBraghetto I have an implementation of dm_control `suite` that is fully based on MJX and will be open-sourced very soon.

sibocw, 2024-10-21:
@kevinzakka Awesome! Looking forward :)