google-deepmind/dm_control #491: Bug in adding distance sensors using mjcf

mohakbhardwaj, 2024-08-28:
Hello, I am using the latest version of `mujoco==3.2.2` and `dm_control==1.0.22` and trying to add distance sensors using MJCF programmatically. I want to add distance sensors only between two bodies and based on the mujoco documentation it seems like I can specify only `body1` and `body2` variables to achieve that (https://mujoco.readthedocs.io/en/latest/XMLreference.html#sensor-distance). However, doing this via mjcf using something like

```
mjcf_model.sensor.add(element_name='distance', name='distance_sensor', body1='body_1', body2='body_2)
```
raises the following `AttributeError: during initialization of attribute 'geom1' of element <distance>: Attribute 'geom1' of element <distance> is required`. I see a similar error if I just specify two geoms. By using dummy values for geom names, there is an error when creating the mujoco physics. Below is a minimal example showing all three failing test cases. 

```
from dm_control import mjcf

mjcf_model = mjcf.RootElement(model='robot')
body_1 = mjcf_model.worldbody.add('body', name='body1')
body_1.add('geom', name='arm', type='capsule', pos=[0, 0, -0.15], size=[0.045, 0.15])

body_2 = mjcf_model.worldbody.add('body', name='body2')
body_2.add('geom', name='arm2', type='capsule', pos=[0, 0, -0.15], size=[0.045, 0.15])

TEST_CASE_NUM = 1

if TEST_CASE_NUM == 1:

    mjcf_model.sensor.add(
        element_name='distance',
        name='distance_sensor',
        cutoff=0.1,
        body1=body_1.name,
        body2=body_2.name,
    )

elif TEST_CASE_NUM == 2:

    mjcf_model.sensor.add(
        element_name='distance',
        name='distance_sensor',
        cutoff=0.1,
        geom1='arm',
        geom2='arm2',
    )

elif TEST_CASE_NUM == 3:

    mjcf_model.sensor.add(
        element_name='distance',
        name='distance_sensor',
        cutoff=0.1,
        geom1='arm',
        geom2='arm2',
        body1=body_1.name,
        body2=body_2.name,
    )

physics = mjcf.Physics.from_mjcf_model(mjcf_model)
```
The bug seems to come from the following file https://github.com/google-deepmind/dm_control/blob/main/dm_control/mjcf/schema.xml#L2647-L2657 that defines both geoms and bodies as required, which does not coincide with mujoco documentation. Is there a way to fix this? Any help would be appreciated. Thanks!

kevinzakka, 2024-08-28:
Hi @mohakbhardwaj, I think I introduced this bug in commit 9c295be579b20317ce7dc1134a8c7ccd6c6faf29. I'll push out a fix but it will only be available in the next release.

mohakbhardwaj, 2024-08-28:
Thanks @kevinzakka ! Building from source solved the issue.

closed 2024-08-28 by yuvaltassa