google-deepmind/dm_control #495: `replicate` does not work in PyMJCF

jonzamora, 2024-09-22:
Hello,

I am trying to load an XML which uses the `replicate` feature from MuJoCo with the `PyMJCF` package, but I get an error.

To reproduce the error, try running the below script:

```python
from dm_control import mjcf

if __name__ == "__main__":
    xml_string = """
    <mujoco model="Particle">
        <worldbody>
            <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5" castshadow="false"/>
            <light diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4"/>
            <geom name="ground" type="plane" size="0 0 1" condim="1"/>
            <replicate count="10" offset=".07 0 0">
                <body pos="-0.35 -0.35 0.65">
                    <joint type="slide" axis="1 0 0" group="3"/>
                    <joint type="slide" axis="0 1 0" group="3"/>
                    <joint type="slide" axis="0 0 1" group="3"/>
                    <geom size="0.025" condim="1" priority="1" rgba="0.8 0.2 0.1 1"/>
                </body>
            </replicate>
        </worldbody>
    </mujoco>
    """
    mjcf_model = mjcf.from_xml_string(xml_string)

    print(mjcf_model)
```

The model loads fine if I drag the XML into the MuJoCo viewer. The error with `PyMJCF` says:

```
KeyError: "Line 7: error while parsing element <replicate>: 'replicate'"
```

Pixxelpulserahul, 2024-09-22:
> Hello,
> 
> I am trying to load an XML which uses the `replicate` feature from MuJoCo with the `PyMJCF` package, but I get an error.
> 
> To reproduce the error, try running the below script:
> 
> ```python
> from dm_control import mjcf
> 
> if __name__ == "__main__":
>     xml_string = """
>     <mujoco model="Particle">
>         <worldbody>
>             <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5" castshadow="false"/>
>             <light diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4"/>
>             <geom name="ground" type="plane" size="0 0 1" condim="1"/>
>             <replicate count="10" offset=".07 0 0">
>                 <body pos="-0.35 -0.35 0.65">
>                     <joint type="slide" axis="1 0 0" group="3"/>
>                     <joint type="slide" axis="0 1 0" group="3"/>
>                     <joint type="slide" axis="0 0 1" group="3"/>
>                     <geom size="0.025" condim="1" priority="1" rgba="0.8 0.2 0.1 1"/>
>                 </body>
>             </replicate>
>         </worldbody>
>     </mujoco>
>     """
>     mjcf_model = mjcf.from_xml_string(xml_string)
> 
>     print(mjcf_model)
> ```
> 
> The model loads fine if I drag the XML into the MuJoCo viewer. The error with `PyMJCF` says:
> 
> ```
> KeyError: "Line 7: error while parsing element <replicate>: 'replicate'"
> ```
> 
> Any help is greatly appreciated

can you show me the file path?

yuvaltassa, 2024-09-22:
replicate is currently not natively supported by PyMJCF. However you can load and save the XML to expand the replicate macro, and load that.

closed 2024-09-22 by yuvaltassa

jonzamora, 2024-09-22:
Hi @yuvaltassa , can you elaborate on what you mean by "expand the replicate macro and load that"?

yuvaltassa, 2024-09-22:
replicate is like a macro. A shorthand for more MJCF. If you load and save an XML with replicate you get the actual MJCF.

Read about meta elements here

https://mujoco.readthedocs.io/en/latest/XMLreference.html#meta-elements