google-deepmind/dm_control #502: How to get started with manipulator control.

gabrielpeixoto-cvai, 2024-11-28:
I have been using Mujoco for some weeks now. I found that dm_control has a simple [Ik function](https://github.com/google-deepmind/dm_control/blob/main/dm_control/utils/inverse_kinematics.py) and the [mjcf module](https://github.com/google-deepmind/dm_control/tree/main/dm_control/mjcf) and I want to try those.

I created an environment where I can control a robot joint using keyboard and the [passive viewer](https://mujoco.readthedocs.io/en/stable/python.html#passive-viewer) for convenience.

Is there a way to achieve a similar setup using the dm_control bindings? 

I figured out that you need to define an environment in order to launch the viewer? Is there a simpler way?

If not, can you point me to the direction in which I should follow to be able to control a manipulator and use the ik function?

For reference, here is my code in mujoco, I basically want to achieve the same thing with dm_control:

```
import mujoco
import numpy as np
from math import pi
from mujoco import viewer

# Load model and initialize data
xml = "universal_robots_ur5e/scene.xml"  # Change to your model's path
model = mujoco.MjModel.from_xml_path(xml)
data = mujoco.MjData(model)
init_q = [-pi/2, 0.0, 0.0, 0.0, 0.0, 0.0]
data.qpos = init_q
mujoco.mj_forward(model, data)

# Define variables
goal = np.array([0.1, 0.1, 0.3])
step_size = 0.5
tol = 0.001
damping = 0.15

# Marker setup
marker_id = 0  # Assuming the first mocap is the marker
data.mocap_pos[marker_id] = goal  # Initialize marker position
mujoco.mj_forward(model, data)
paused = False

def key_callback(keycode):
    print(f"Char: {chr(keycode)} - keycode: {keycode}")
    increment = 0.025
    print(f"Current goal: {goal}")
    if chr(keycode) == ' ':
      paused = not paused    
    elif keycode == 265:
        # x up
        print(f"{goal[0]}->{goal[0] + increment}")
        goal[0] = goal[0] + increment
    elif keycode == 264:
        # x down
        goal[0] = goal[0] - increment
    elif keycode == 263:
        # y up
        goal[1] = goal[1] + increment
    elif keycode == 262:
        # y down
        goal[1] = goal[1] - increment
    elif keycode == 266:
        # z up page up
        goal[2] = goal[2] + increment
    elif keycode == 267:
        # z down pagedown
        goal[2] = goal[2] - increment
    print(f"New goal = {goal}")
    data.mocap_pos[marker_id] = goal
    mujoco.mj_forward(model, data)
        

# Function to calculate inverse kinematics (IK)
def calculate_ik(goal, step_size, tol, damping):
    jacp = np.zeros((3, model.nv))
    error = np.subtract(goal, data.body("link_6").xpos)

    while np.linalg.norm(error) >= tol:
        mujoco.mj_jac(model, data, jacp, np.zeros((3, model.nv)), goal, data.body("link_6").id)
        I = np.identity(model.nv)
        j_inv = np.linalg.pinv(jacp.T @ jacp + damping * I) @ jacp.T
        delta_q = j_inv @ error

        # Update joint positions
        data.qpos += step_size * delta_q
        mujoco.mj_step(model, data)
        error = np.subtract(goal, data.body("link_6").xpos)

def check_joint_limits(q):
    """Check if the joints is under or above its limits"""
    for i in range(len(q)):
        q[i] = max(model.jnt_range[i][0], min(q[i], model.jnt_range[i][1]))

jacp = np.zeros((3, model.nv)) #translation jacobian
jacr = np.zeros((3, model.nv)) #rotational jacobian
step_size = 1
tol = 1e-14
alpha = 0.5
damping = 0.15

#Get error.
end_effector_id = model.body('wrist_3_link').id #"End-effector we wish to control.
current_pose = data.body(end_effector_id).xpos #Current pose
error = np.subtract(goal, current_pose) #Init Error

# Initialize viewer
with mujoco.viewer.launch_passive(model, data, key_callback=key_callback) as viewer:

    # Main rendering and simulation loop
    try:
        while viewer.is_running():
            with viewer.lock():
                viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True
                viewer.opt.flags[mujoco.mjtFrame.mjFRAME_GEOM] = True

            
            if (np.linalg.norm(error) >= tol):
                mujoco.mj_jac(model, data, jacp, jacr, goal, end_effector_id)
                #Calculate delta of joint q
                n = jacp.shape[1]
                I = np.identity(n)
                product = jacp.T @ jacp + damping * I

                if np.isclose(np.linalg.det(product), 0):
                    j_inv = np.linalg.pinv(product) @ jacp.T
                else:
                    j_inv = np.linalg.inv(product) @ jacp.T

                delta_q = j_inv @ error

                #Compute next step
                q = data.qpos.copy()
                q += step_size * delta_q

                #Check limits
                check_joint_limits(data.qpos)
                # print(error)
                #Set control signal
                q[5] = -q[5]
                data.ctrl = q 
                print(q)
                mujoco.mj_forward(model, data)
                #Step the simulation.
                mujoco.mj_step(model, data)
                # error = np.subtract(goal, data.body(end_effector_id).xpos)
            error = np.subtract(goal, data.body(end_effector_id).xpos)
            viewer.sync()
    finally:
        viewer.close()
```

yuvaltassa, 2024-12-02:
Why do you want to use dm_control rather than native MuJoCo?

gabrielpeixoto-cvai, 2024-12-03:
@yuvaltassa 
Thanks for the response. Indeed I was thinking about this after I created this issue.
The biggest plus for my use case is the MJCF module that allows to create dynamic scenes. I did not experiment with manipualtors yet, but I want to create obstacles in random positions, for example, that would be useful.

closed 2024-12-05 by gabrielpeixoto-cvai

gabrielpeixoto-cvai, 2024-12-10:
@yuvaltassa Thank you very much!
I was not aware that native mujoco had this feature! I will check it soon!