google-deepmind/dm_control #503: How to adjust the size of bolt and nut from the plugin?

figsama, 2024-12-31:
Hello! I'm an undergraduate student trying to learn mujoco. I'm trying to use UR5E with a gripper to screw a nut, only to find that the manipulator is small and the nut is super big.  the bolt and nut are from the mujoco plugin, I tried a lot but I cannot change the size of the nut. Could you please tell me what can I do now? Thanks a lot!

yuvaltassa, 2025-01-02:
You read this?

https://github.com/google-deepmind/mujoco/blob/main/plugin/sdf/README.md

It's referenced here:

https://mujoco.readthedocs.io/en/latest/programming/extension.html#sdf

figsama, 2025-01-05:
> You read this?
> 
> https://github.com/google-deepmind/mujoco/blob/main/plugin/sdf/README.md
> 
> It's referenced here:
> 
> https://mujoco.readthedocs.io/en/latest/programming/extension.html#sdf

Thank you for your kind reply. I have read these documents, but I find that I can only adjust the radius of bolt and nut by changing the "radius" parameter, not the proportional reducing the size. By adjusting the radius, the thread part can be thickened or thinned, but the size of the nut remains unchanged. I'm not sure if I need to rewrite the head files to do that. Could you told me what to do?

figsama, 2025-01-08:
As a supplement, the nut and bolt look as what is shown in this picture:
[image]

> You read this?
> 
> https://github.com/google-deepmind/mujoco/blob/main/plugin/sdf/README.md
> 
> It's referenced here:
> 
> https://mujoco.readthedocs.io/en/latest/programming/extension.html#sdf