google-deepmind/dm_control #505: Include tags are not properly handled.

QuantuMope, 2025-01-06:
Hello, I'm trying to use `<include>` tags to modularize my xml files, but it seems that the `dm_control/mjcf/parser.py` does not handled nested includes.

Below is the main xml
```xml
<mujoco>

  <asset>
    <mesh name="link_camera_mount" file="meshes/zed_mini_camera_mount_26deg.stl" scale="0.01 0.01 0.01"/>
  </asset>

  <default>
    <default class="visual">
      <geom contype="0" conaffinity="0" type="mesh"/>
    </default>
  </default>

  <worldbody>
    <body name="camera">
      <body name="link_camera_mount">
        <inertial pos="-1.07532924e-02 -4.53753013e-06  2.89950897e-02"
                  quat="0.69670638  0.69711755 -0.11961327  0.11966623"
                  mass="0.01"
                  diaginertia="7.00807202e-06 3.96812607e-06 3.35458862e-06"/>
        <geom class="visual" type="mesh" mesh="link_camera_mount"/>
      </body>

      <include file="zed_mini.xml" />

    </body>
  </worldbody>

</mujoco>
```
Now, here is the referenced "zed_mini.xml"

```xml
<mujoco model="zed_mini">
    <body name="zed_mini" pos="0.0014 0.0087 0.04437" quat="0.22361 0 -0.97468 0">
      <geom name="zed_mini" type="box" size="0.0151 0.0626 0.0133" rgba="1 1 1 0.5" mass="0.060"/>

      <camera name="lens" mode="fixed" pos="-0.0151 -0.04025 0" fovy="54" ipd="0.063" xyaxes="0 -1 0 0 0 -1"/>
      <camera name="lens_left" mode="fixed" pos="-0.0151 0.02275 0" fovy="54" ipd="0.063" xyaxes="0 -1 0 0 0 -1"/>
    </body>
</mujoco>
```

This results in the following stacktrace
```
  File "/home/asjchoi/Desktop/Hobot/hobot/environments/mj_models/cameras/camera.py", line 78, in __init__
    mjcf_element = load_mjcf_from_path(xml_path)
  File "/home/asjchoi/Desktop/Hobot/hobot/environments/mj_models/utils.py", line 34, in load_mjcf_from_path
    return mjcf.from_path(str(path))
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 105, in from_path
    return _parse(xml_root, escape_separators,
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 172, in _parse
    _parse_children(xml_root, mjcf_root, escape_separators)
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 222, in _parse_children
    _parse_children(xml_child, mjcf_child, escape_separators)
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 222, in _parse_children
    _parse_children(xml_child, mjcf_child, escape_separators)
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 219, in _parse_children
    raise err_type(  # pylint: disable=raise-missing-from
  File "/home/asjchoi/Desktop/alf/venv/lib/python3.10/site-packages/dm_control/mjcf/parser.py", line 200, in _parse_children
    child_spec = mjcf_element.spec.children[xml_child.tag]
KeyError: "Line 41: error while parsing element <include>: 'include'"
```

Saw that this issue was opened previously #240 but perhaps never handled?

Wondering if this is a lack of feature or if I am setting up the include correctly. My convention seems to align with MuJoCo's examples.

yuvaltassa, 2025-01-10:
PyMJCF is (slowly) being phased out.
Do you want to try using the new [native attach](https://mujoco.readthedocs.io/en/latest/python.html#attachment) functionality for your needs?

QuantuMope, 2025-01-10:
> PyMJCF is (slowly) being phased out. Do you want to try using the new [native attach](https://mujoco.readthedocs.io/en/latest/python.html#attachment) functionality for your needs?

I see. Though, this is a more programmatic solution to attachments? Is there then no way to handle modular attachments at the XML level?

yuvaltassa, 2025-01-10:
https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-attach

which is clearly mentioned here:

https://mujoco.readthedocs.io/en/latest/programming/modeledit.html#attachment

closed 2025-01-10 by yuvaltassa

QuantuMope, 2025-01-10:
@yuvaltassa Thanks for the prompt response!