google-deepmind/dm_control #508: `springlength` does not support two values

ekchapman, 2025-01-15:
Mujoco supports a 2-value input for `springlength` ([see docs](https://mujoco.readthedocs.io/en/stable/XMLreference.html#tendon-spatial-springlength)), but I'm getting an error when using `dm_control.mjcf`. I thought it was fixed per https://github.com/google-deepmind/dm_control/pull/358, but I'm still getting an error. Is this a bug? :thinking:

To reproduce:
```bash
pyenv install 3.10.12
pyenv virtualenv 3.10.12 springlength-bug
pyenv activate springlength-bug
pip install dm-control==1.0.26
python3 repro.py
pyenv deactivate
pyenv uninstall springlength-bug
```

```python
# repro.py

from dm_control import mjcf

root = mjcf.RootElement()

types = ['spatial', 'fixed']
springlengths = [[0.0, 1.2], "0.0 1.2"]

for tendon_type in types:
    for springlength in springlengths:
        try:
            tendon = root.tendon.add(
                    tendon_type,
                    springlength=springlength
            )
        except Exception as e:
            print(f"Error: {e}")

```

```
Error: during initialization of attribute 'springlength' of element <spatial>: float() argument must be a string or a real number, not 'list'
Error: during initialization of attribute 'springlength' of element <spatial>: Expect a float value: got 0.0 1.2
Error: during initialization of attribute 'springlength' of element <fixed>: float() argument must be a string or a real number, not 'list'
Error: during initialization of attribute 'springlength' of element <fixed>: Expect a float value: got 0.0 1.2
```

yuvaltassa, 2025-01-15:
Can you inspect `mjcf/schema.xml` and see if you detect a mismatch?

Also, unless you already have lots of code in PyMJCF, I'd recommend taking a look and the new native model editing capabillities.

ekchapman, 2025-01-15:
`mjcf/schema.xml` appears correct; it has this text under spatial and fixed tendons:
```
<attribute name="springlength" type="float" array_size="2"/>
```

To reproduce:
Open the `.xml` file adjacent to this one:
```
python3 -c "from dm_control import mjcf; print(mjcf.schema.__file__)"
/home/evan/.pyenv/versions/springlength-bug/lib/python3.10/site-packages/dm_control/mjcf/schema.py
```

> I'd recommend taking a look and the new native model editing capabilities.
Totally! I've been really excited about this and many other recent updates — I'm trying it out now and it's really nice. I will migrate, perhaps sooner if this is indeed a bug.

yuvaltassa, 2025-01-15:
Ah, I think I see the bug. Instead of
```xml
<attribute name="springlength" type="float" array_size="2"/>
```
it should be
```xml
<attribute name="springlength" type="array" array_type="float" array_size="2"/>
```
Can you try and change locally and see if that fixes it?

ekchapman, 2025-01-15:
Yes, that fixes it!

commit b2484eee (2025-01-16): Fix schema definition of `springlength`, fixes #508

PiperOrigin-RevId: 716236616
Change-Id: I0a37fc0737ff577bab870b0fe0a6652236ac4fce
closed 2025-01-16 by copybara-service[bot]