google-deepmind/dm_control #518: Mesh rendering as sphere?

tonitonitonitoni, 2025-05-19:
Hi, I am trying to load a mesh from a file using the PyMJCF API. But it renders all my meshes as spheres.  Am I doing something wrong? Here is my code.  
Replace with the STL of your choice or save this to cube.obj - they all render as spherical.  
**cube.obj:** 
```
    v -1 -1  1
    v  1 -1  1
    v -1  1  1
    v  1  1  1
    v -1  1 -1
    v  1  1 -1
    v -1 -1 -1
    v  1 -1 -1
```

**mesh_test.ipynb:**
```
stl_file = 'cube.obj'
root = mjcf.RootElement()
getattr(root.visual, "global").offwidth = 2560
getattr(root.visual, "global").offheight = 1440
    
root.asset.add('mesh', name='base', file=stl_file)
root.worldbody.add('geom', mesh='base', pos=[0,0,0], mass=10)

model = mujoco.MjModel.from_xml_string(root.to_xml_string(), root.get_assets())
data = mujoco.MjData(model)
mujoco.mj_forward(model, data)
with mujoco.Renderer(model) as renderer:
    renderer.update_scene(data)
    pixels = renderer.render()
media.show_image(pixels)
```

Thanks so much for your attention to this matter.

tonitonitonitoni, 2025-05-19:
Thank you SO MUCH!! it worked beautifully!
As does MINK! Thank you **SO MUCH**

yuvaltassa, 2025-05-19:
To clarify what is actually happening here, the code was missing `type="mesh"`. When you specify a mesh together with a **primitive geom**, you are asking MuJoCo to approximate the mesh as a primitive. Since the default geom type is a sphere, you got spheres.